drahnr/oregano #144: add to macports

reubano, 2015-09-11:
Could you please add this to [macports](https://guide.macports.org/chunked/development.creating-portfile.html)?

reubano, 2015-09-11:
Or better yet, take over the [current port](https://trac.macports.org/browser/trunk/dports/gnome/oregano).

drahnr, 2015-09-16:
I do not own a mac, I sure have a few colleagues with a mac but I definitely need guidance on doing so.

reubano, 2015-09-16:
I've never built a macport myself, so I can't be of much help. However, the above links should be able to point you in the right direction.

drahnr, 2015-09-16:
I need you as my guinea pig for the mac ports I hack together :)

I could maybe do it on Friday, but not promises made.

drahnr, 2015-09-18:
Just because you never did it does not mean that you can not do it.

drahnr, 2015-09-21:
To be honest, this is effort that is currently not worth it.

If you can provide a PR I'll happily merge them.

closed 2015-09-21 by drahnr