drahnr/oregano #152: Included Portfile for MacPorts fails to build on MacPorts as distributed

dbevans, 2015-12-29:
Using your Portfile as a starting point, I've updated the official MacPorts version of oregano to the latest 0.83.2.1 release from this repo.  Since this port now builds using waf, I've used our waf 1.0 port group to help simplify the port and corrected the maintainer syntax and other minor issues.  Our version of the port also allows building of the app with either the gtk3 x11 or quartz backends.  

See MacPorts ticket https://trac.macports.org/ticket/49412 and 
changeset https://trac.macports.org/changeset/144007 

for more details and feel free to open a ticket at https://trac.macports.org if you have any updates or other changes you would like to make. You may  also want to update your distributed version of the Portfile to more closely match ours.

drahnr, 2016-01-03:
The MacPorts file was just a draft, I do not own a Mac nor do I intend to buy one and thus can not test the MacPorts file.

If you fix this I'll happily apply your patch :)

dbevans, 2016-01-04:
Probably easiest to just replace it.  You can find the current version of the port here:
http://svn.macports.org/repository/macports/trunk/dports/gnome/oregano/Portfile

drahnr, 2016-01-04:
Did you test that? With the the latest release?

dbevans, 2016-01-04:
Yes, this is the official MacPorts release for oregano 0.83.2.1.  Tested successfully on MacPorts buildbots for OS X 10.6, 10.8, 10.9, 10.10.  Tested locally on 10.11 (El Capitan).

drahnr, 2016-01-04:
Fixed in b3338ccbddee3c64b65f3f6c52f61a329c681d74

closed 2016-01-04 by drahnr