drahnr/oregano #154: builddeps should prompt before installing/updating.

ghedberg, 2016-04-04:
### Steps to reproduce
1. run builddeps

### Expected behaviour
Tell us what should happen

I should be prompted as to what the installer is intending to install before installing.

### Actual behaviour

The automatic update and install of packages installed a buggy kernel driver on my system causing X to stop working (after reboot).

### Configuration
**Operating system**:  fedora 23 on a HP laptop (i915 is buggy as hell)
```
cat /etc/os-release |grep PRETTY_NAME
```

Essentially I really think that both updating and installing packages with -y and no user prompt is a really bad idea. I guess I should have read the code before running, but I think that it is standard practice to prompt the user to choose if they want to proceed.

The update installed a buggy intel X driver which a few days later when I rebooted, X refused to detect my screens and caused me a couple of hours of backtracking to find what happened (which is not fun to do without a working screen).

drahnr, 2016-04-05:
Agreed, I did not consider those use cases enough.

closed 2016-04-05 by drahnr

ghedberg, 2016-04-05:
Looks good to me. Thanks

commit 99633651 (2016-04-05): ci: Prevent regular developers from auto updating, closes #154.
commit b760789f (2016-04-05): ci: Prevent regular developers from auto updating, closes #154.
commit ec1aa8e2 (2016-04-05): ci: Only ci shall do a system update, closes #154.