drahnr/oregano #161: Lack of people and public interest

drahnr, 2017-03-08:
Since I am entangled in more than just one new endeavor, I am looking for someone to finish the gtk3 adjustments or for contributors and potential future maintainers.

If nobody steps up, this project has to be considered dead.

junaidnaseer, 2017-11-08:
Hi,

Are you still looking for a maintainer?

drahnr, 2017-11-08:
Yes, I am looking for someone who knows. Do you have experience with Gtk+3.x?

drahnr, 2021-03-21:
Last call. Archiving on March 28th.

drahnr, 2021-03-22:
@wereii && @surprized if you want to make this project live on, I am up to maintaining the CI + packaging, as well as mentoring. But somebody has to do the actual "product" development :upside_down_face:

surprized, 2021-03-22:
Really wish I could help, but I don't think I have sufficient coding knowledge or the time to help right now. :(

wereii, 2021-03-22:
I am afraid but I have minimal exposure to the usage of this software - I am way more of an software guy than electronics. Would love to keep this afloat but I don't think I am the right guy.