drahnr/oregano #164: Crash at simulation-run-time for specific Oregano file

Michi1994, 2017-05-15:
### Steps to reproduce
1. open file [crash.txt](https://github.com/drahnr/oregano/files/1001286/crash.txt) with Oregano
2. press "run simulation"

### Expected behaviour
no crash (but maybe an error)

### Actual behaviour
crash
NULL pointer at
**file** ngspice-analysis.c
**function** void parse_transient_analysis (OreganoNgSpice *ngspice, gchar *tmp)
**variable** gchar **variables

The variable is set to NULL because
function gchar **get_variables (const gchar *str, gint *count)
is printing the warning "NO COLUMNS FOUND" and returns NULL.

### Configuration
**Operating system**:
```
$ cat /etc/os-release |grep PRETTY_NAME
Ubuntu 16.10
```

**Versions**
```
$ oregano --version
Oregano 0.83.3
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster
$ ngspice --version
ngspice compiled from ngspice revision 26
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project
```

Michi1994, 2017-05-15:
I set R2 to 8 and there was no crash. I watched the output of ngspice which is stored at /tmp/netlist.lst and I compared the two outputs. This is possible because the crash happens at file-parsing-time so ngspice did its job already and stored the output to /tmp/netlist.lst. As you can see, in [netlist (R2=9).lst](https://github.com/drahnr/oregano/files/1001484/netlist.R2.9.txt) the line with "No. of Data Rows" is missing. Furthermore the end simulation time of [netlist (R2=9).lst](https://github.com/drahnr/oregano/files/1001484/netlist.R2.9.txt) is not the expected end simulation time. This could mean that ngspice did not end the simulation and there was an error. I will check now if the ngspice-error-recognition-algorithm of Oregano is correct.

[netlist (R2=8).lst](https://github.com/drahnr/oregano/files/1001485/netlist.R2.8.txt)
[netlist (R2=9).lst](https://github.com/drahnr/oregano/files/1001484/netlist.R2.9.txt)

drahnr, 2017-05-15:
Yep, that would be great!

Michi1994, 2017-05-15:
If I run the simulation (to write the netlist to "/tmp/netlist.tmp"), let it crash, then do
```
$ ngspice -b netlist.tmp
$ echo $?
```
the output is
```1```
meaning that the exit code could be used to check whether ngspice succeeded or not.
I think that Oregano tries to read the exit code but does not succeed and here is why I think this is the case:
At
**file** ngspice.c
**function** static void ngspice_start (OreganoEngine *self)
there is attached the
**function** static void ngspice_watch_cb (GPid pid, gint status, OreganoNgSpice *ngspice)
to the ngspice subprocess as a watch function. I think, the variable status should be the exit status of the ngspice subprocess. But it's not. There is occurring a warning message
```
GChildWatchSource: Exit status of a child process was requested but ECHILD was received by waitpid(). Most likely the process is ignoring SIGCHLD, or some other thread is invoking waitpid() with a nonpositive first argument; either behavior can break applications that use g_child_watch_add()/g_spawn_sync() either directly or indirectly.
```
what indicates that something went wrong at getting the exit code.

Michi1994, 2017-05-15:
Now I know why the status was not the real status and why the warning message appeared.
It was because at
**file** ngspice.c
**function** static void ngspice_start (OreganoEngine *self)
the
**function** g_child_watch_add_full
has been called false. The priority and pid parameters have been interchanged. Now that I interchanged it correctly, the status seems to be the real status.

Michi1994, 2017-05-15:
If I run the simulation with a working file (i.e. R2=9), the status of
**function** static void ngspice_watch_cb (GPid pid, gint status, OreganoNgSpice *ngspice)
is 0. If I run the simulation with R2=8, the status is 256 (0x100). Now the question is, why are only the first 7 least significant bits of status relevant?
```
if (((status) & 0x7f) == 0) {
...
}
```

drahnr, 2017-05-16:
I think the common assumption is that the return value is exactly one byte in size, where practically it is most likely an 32bit integer.

http://stackoverflow.com/questions/8082953/what-is-the-valid-range-for-program-return-value-in-linux-bash

Michi1994, 2017-05-17:
I got this out of `man waitpid`
```
      WIFEXITED(wstatus)
             returns true if the child terminated normally, that is, by call‐
             ing exit(3) or _exit(2), or by returning from main().
```
so ngspice ends normally (maybe by returning from main()), but the return value is not 0, because there was a simulation error such as a not converging algorithm at a time less than the expected end time. So this issue can't be solved completely by Oregano but has to be solved by ngspice. ngspice is producing normal output despite there was an error. What we can do is plotting as much as we can and popping up a warning.

Michi1994, 2017-05-17:
I'm done with this issue. I did my best. I have to go on.

drahnr, 2017-05-17:
I will try to look into it this weekend

Michi1994, 2017-05-23:
You merged the changes already, as far as I can see. So you can close this issue?
I requested a new proposal to the behavior of the ngspice error handling in pull request #171. If you accept the proposal, there will not be printed the solution as far as possible but the ngspice output file will be printed in the log view instead.

drahnr, 2017-05-23:
No not yet, I did not get around to look into the details of it just yet.

Michi1994, 2017-06-16:
The given file does not produce any errors any more. This issue is obsolete.

closed 2017-06-16 by drahnr