drahnr/oregano #168: Node referencing algorithm has to imrove

Michi1994, 2017-05-16:
### Steps to reproduce
1. define a "non-linear current source" (see: [ngspice doc](http://ngspice.sourceforge.net/docs/ngspice-manual.pdf) 5.3 Gxxxx: non-linear current source*) in default.oreglib like this:
```
    <ogo:part>
      <ogo:name>nlcs</ogo:name>
      <ogo:symbol>resistor</ogo:symbol>
      <ogo:description>non-linear current source</ogo:description>
      <ogo:labels>
        <ogo:label>
          <ogo:name>Reference designator</ogo:name>
          <ogo:text>@refdes</ogo:text>
          <ogo:position>(15 0)</ogo:position>
          <ogo:modify>yes</ogo:modify>
        </ogo:label>
      </ogo:labels>

      <ogo:properties>
        <ogo:property>
          <ogo:name>Refdes</ogo:name>
          <ogo:value>G</ogo:value>
        </ogo:property>
        <ogo:property>
          <ogo:name>Cur</ogo:name>
          <ogo:value>'expr'</ogo:value>
        </ogo:property>
        <ogo:property>
          <ogo:name>Template</ogo:name>
          <ogo:value>@refdes %1 %2 cur=@cur</ogo:value>
        </ogo:property>
      </ogo:properties>
    </ogo:part>
```
2. open [tryNLCS.oregano](https://github.com/drahnr/oregano/files/1005366/tryNLCS.oregano.txt)
3. run simulation

### Expected behaviour
The defined current curve in parameter "Cur" indicates that the "non-linear current source" is a simple resistor with a resistance of 1k, so the simulation should run without problems.

### Actual behaviour
The netlist generation fails because the %-replace-algorithm is too stupid.

### Configuration
**Operating system**:
```
$ cat /etc/os-release |grep PRETTY_NAME
PRETTY_NAME="Ubuntu 16.10"
```

**Versions**
```
$ oregano --version
Oregano 0.83.3
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster
$ ngspice --version
ngspice compiled from ngspice revision 26
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project
```

Michi1994, 2017-05-16:
The %-replace-algorithm can be found at
**file** netlist-helper.c
**function** void netlist_helper_create (Schematic *sm, Netlist *out, GError **error)
.

drahnr, 2017-05-17:
Feel free to rewrite that whole interface, it is pretty simple in it's nature and I never touched it tbh.

Michi1994, 2017-05-17:
solution
========

new syntax
----------

**definition: %\<number of connection\>**
Because the `<ogo:connection>` tag of oreglib files do not have a designator, the only way to reference a connection is by its number. So the designator is implicitly defined by the occurrence-number in the oreglib file.

**example (oreglib)**
The following example is a "non-linear current source" but because of the simple definition of the "Cur"-property, it is a simple resistor with resistance of 1k (as long as the "Cur"-property will not be changed by the user).
```
    <ogo:part>
      <ogo:name>nlcs</ogo:name>
      <ogo:symbol>resistor</ogo:symbol>
      <ogo:description>non-linear current source</ogo:description>
      <ogo:labels>
        <ogo:label>
          <ogo:name>Reference designator</ogo:name>
          <ogo:text>@refdes</ogo:text>
          <ogo:position>(15 0)</ogo:position>
          <ogo:modify>yes</ogo:modify>
        </ogo:label>
      </ogo:labels>

      <ogo:properties>
        <ogo:property>
          <ogo:name>Refdes</ogo:name>
          <ogo:value>G</ogo:value>
        </ogo:property>
        <ogo:property>
          <ogo:name>Cur</ogo:name>
          <ogo:value>'0.001*(V(%1)-V(%2))'</ogo:value>
        </ogo:property>
        <ogo:property>
          <ogo:name>Template</ogo:name>
          <ogo:value>@refdes %1 %2 cur=@cur</ogo:value>
        </ogo:property>
      </ogo:properties>
    </ogo:part>
```

**possible advantages if no designators will be definable for connections**
- defining designators for each connection is not fun and can be left out
- converting the existing libraries to the new concept is simple
- simple concept

**possible disadvantages if no designators will be definable for connections**
- The template property would be more readable for humans.
- The designators defined in the ngspice/gnucap documentation could be used and it would be more clear which designator belongs to a specific connection.

**problems if designators will be definable for connections**
- What if the designators of gnucap are not the same as the designators of ngspice?
- Which characters should be allowed in a designator so that the parsing does not fail at specific designator strings?
- Which character sequence should mark the beginning of a connection reference and which character sequence should mark the end of a connection reference (currently: "%" is the beginning character sequence and "\<no digit\>" is the end character sequence)?

backward compatability
----------------------

Old oregano files will not work anymore because there have been used arbitrary connection designators in the past very often.

**possible solutions**
- Doing nothing and letting happen a crash of the program and the user has no idea what's going on. After several hours of debugging and searching in documentation for solution he finds out that the syntax has changed and he has to convert his file by hand. _This is the solution I prefer, because I don't think that there are so many users that use Oregano._
- Offering an external converting program so that the user does not have to convert by hand (problem with "several hours of debugging" not solved).
- Implementing a deprecated-version-of-file-recognition-algorithm that checks if a file (which the user wants to open) has a deprecated syntax. Throw a warning.
- Implementing a deprecated-version-of-file-recognition-algorithm and converting the file automatically.

drahnr, 2017-05-17:
I think you are right that not many use oregano, but those who do have lots of legacy schematic hanging around. Not sure if Option number 4 is not a better solution.

Michi1994, 2017-05-17:
Ok. As you wish. I already wrote a small piece of Java code that helped me converting the old libraries automatically. The algorithm is working without any errors. Tomorrow I will implement a deprecated-version-of-file-recognition-algorithm and I will transform the piece of Java code to a piece of C code.

Michi1994, 2017-05-18:
Today I tried to work with the old ugly+bad code at
**file** part-property.c
**function** char *part_property_expand_macros (Part *part, char *string)
, but the ugly+bad code is also buggy. Now I know that I have to write my own macro expansion code.

I also implemented the deprecated-file-recognition-algorithm.
I thought about 2 possible concepts:
1. Analyzing the content and searching possible structures that have to be converted.
2. Defining a new tag `<ogo:version>` that describes which version of Oregano has created that file.

I think the second solution is better because in the future, we will change the layout of the files, adding features and so on. And if the version is documented, we can define a conversion algorithm for each version step so that the latest version of Oregano can read and handle all files of all previous versions. So I chose option number 2 and implemented it already. It's working.

drahnr, 2017-05-19:
The second one is preferable.

Yep, it always has been ugly and buggy.  If you rewrite it, I strongly recommend using regular expressions https://developer.gnome.org/glib/stable/glib-regex-syntax.html and writing testcases for it - maybe based on the existing oreglib files.

Michi1994, 2017-05-19:
Yeah. I will use regex but I have to use it piecewise/iterative because of the "separators". The opening separator decides which separator has to be used for the closing separator (it is the same).

How do I write testcases? I know how to write testcases for Java code, but not C code.

The oreglib files are too primitive (and also buggy). They do not use node references outside of `<ogo:template>` and they are using only macro `@`. I have already written some complex parts for manual testing.

drahnr, 2017-05-19:
You should be able to do it in a single line if you use backreferences, see `\ddd` in the docs.

Under `tests` is currently one file, I'd recommend to extend that one with your tests, right now this is only very primitive stuff. It will be built automatically whenever you build.

Michi1994, 2017-05-19:
Backreference is not documented. How to do it?

Michi1994, 2017-05-19:
How can I access the version of Oregano?

Michi1994, 2017-05-19:
The current conversion algorithm is not so simple, because in this version of the algorithm I did not know that back referencing in regex is possible.

drahnr, 2017-05-19:
`VERSION` is defined by the buildsystem which generates `config.h` which is included in `oregano.c` (I think)

Michi1994, 2017-05-19:
No. It is defined in the command line when invoking `gcc` directly as "external macro".

drahnr, 2017-05-20:
Oh, I think th config.h was just introduced in the other PR.

Anyways, it is initially defined in the `wscript` and the passed on.

drahnr, 2017-05-20:
```
#ifndef VERSION
#define VERSION "unknown"
#endif
```
should be a short term workaround

Michi1994, 2017-05-20:
They are all stored in `conf.env.DEFINES`. I will write an ant script or something that executes waf, wscript will print the list `conf.env.DEFINES` to a file and the ant script will then adjust the **file** .cproject of the eclipse project so that the eclipse indexer knows the values of the global macros. That's my plan.

Michi1994, 2017-05-20:
Are there any g assert functions that compare files?

Michi1994, 2017-05-20:
Is there an Oregano API? I can't test my
**function** `void update_schematic(Schematic *sm)`
because it uses a `Schematic` and building a `Schematic` without a full running Oregano seems impossible. Maybe I can test
**function** `void part_property_convert_connection_designators(Part *part, char **prop, int *node_ctr)`
(the real conversion algorithm) because `Part` seems easier to build (and hopefully without a full running Oregano).

Michi1994, 2017-05-20:
I have written and pushed some tests with much code coverage. Now I will try to make the code more readable by using more regex.

Michi1994, 2017-05-20:
I increased the version of Oregano from
0.83.3
to
0.84.0
because I don't know how the Oregano versioning is handled but I think:

- major changes in background or GUI will increase the first number by 1
- minor changes in background or GUI will increase the second number by 1
- bugfixes will increase the third number by 1

Is this correct?

Michi1994, 2017-05-20:
I'm done with using regex. I could use more regex but this will make the code more unreadable. I think the current mixture of regex code and C code is perfect. I'm done with this issue.

drahnr, 2017-05-21:
There is a generator plugin for eclipse projects if I am not mistaken, but I am not a 100% sure.

You could create an oregano schematic (any that is sufficient for the test) and either add it to the test or example dir or copy verbatim into the C file and use that as source. I recommend the file one.

Bump to 0.84.0 is reasonable.

---

I'll try to review this tomorrow.

closed 2017-05-22 by drahnr