drahnr/oregano #172: Oregano - Plot: Resolution has to improve

Michi1994, 2017-05-24:
### Steps to reproduce
1. open [tryResolution.oregano](https://github.com/drahnr/oregano/files/1025268/tryResolution.oregano.txt)
2. run simultation
3. int the Oregano - Plot, zoom in at 0.24ms to view the oscillations

### Expected behaviour
each number on the time axis should not be equal

### Actual behaviour
all numbers on the time axis are equal (0.24ms)

### Configuration
**Operating system**:
```
$ cat /etc/os-release |grep PRETTY_NAME
PRETTY_NAME="Ubuntu 16.10"
```

**Versions**
```
$ oregano --version
Oregano 0.84.1
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster
$ ngspice --version
ngspice compiled from ngspice revision 26
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project
```

drahnr, 2017-05-24:
Do not invest any work on this other than integrating goatplot!

Michi1994, 2017-05-24:
I thought about looking for plotting packages before I began to invest work in it. But I thought also that searching for and integrating another package means more work than adding a small beauty feature. So you searched already and found goatplot. That's great. Next time I will begin integrating goatplot as an alternative (user decides which plot window should be used).

Michi1994, 2017-05-24:
I mean: you wrote goatplot, wow!

commit 59ba382f (2017-05-25): Merge pull request #173 from Michi1994/issue172

plot: adaptive axis number precision, fixed transformation bugs, reset view feature, closes #172
closed 2017-05-25 by drahnr