drahnr/oregano #182: Tests freeze in CI env

drahnr, 2017-06-17:
### Steps to reproduce
1. Just commit something
2. check the compile step `https://ci.spearow.io`

### Expected behaviour
Succceeds

### Actual behaviour
Spins forever

drahnr, 2017-06-17:
Maybe adding a timeout to all `GMainLoop`s running `g_main_loop_run(...)` would be enough

Michi1994, 2017-06-17:
I don't think it's a loop problem. I carefully coded them so that they terminate normally by a chain of events.

I think it's a path resolving problem. I think the system can't find the path to the `expected.txt` file because there appears no `/oregano` string in the path (instead there is `/git-clone-resource`). I think the problem occurs in `static gchar* test_engine_ngspice_get_test_dir_base()`.

Michi1994, 2017-06-17:
How do I commit something?

Michi1994, 2017-06-17:
Why are the tests not executed in [https://ci.spearow.io/teams/main/pipelines/oregano/jobs/compile-pr/builds/54](https://ci.spearow.io/teams/main/pipelines/oregano/jobs/compile-pr/builds/54)?

drahnr, 2017-06-17:
I'll fix that

drahnr, 2017-06-17:
I fixed the CI to always execute the tests.

The fix did not change anything.
It is not a path problem, the execution path changed to `./build/src/microtests`.
I think it would make senso to make those files includes (or `GResouces`, but that would require some more work and adjustment of the buildsystem which I am not exactly sure is worth it)

drahnr, 2017-06-17:
Changes to the CI are tracked in this git repository but I have to manually push them.

drahnr, 2017-06-17:
https://ci.spearow.io/teams/main/pipelines/oregano/jobs/compile/builds/108

commit 6f83f869 (2017-06-17): repair ngspice watcher test, closes #182
commit e527a3df (2017-06-17): Merge pull request #184 from Michi1994/issue182

repair ngspice watcher test, closes #182
commit 80a04427 (2017-06-17): ci: install ngspice and gnucap, fixes #182
commit 7707787e (2017-06-18): repair ngspice watcher test, closes #182
closed 2017-06-17 by drahnr