drahnr/oregano #183: Tests fail locally

drahnr, 2017-06-17:
### Steps to reproduce
1. git clone ... && git checkout b29e8dbf080b740bd49e85d5b9351dfc3a0737f2
2. ./waf configure build
3. ./waf

### Expected behaviour
Success, exit 0

### Actual behaviour
```
/home/bernhard/Projects/oregano/build/src/microtests                                                  10:00:03   1 
/core/coords: OK
/core/engine: OK
/core/model/nodestore: OK
/core/model/wire/intersection: OK
/core/model/wire/tcrossing: OK
/core/model/part-property/update_connection_designators/basic: OK
/core/model/part-property/update_connection_designators/AT: OK
/core/model/part-property/update_connection_designators/AMPERSAND_TRUE: OK
/core/model/part-property/update_connection_designators/AMPERSAND_FALSE: OK
/core/model/part-property/update_connection_designators/QUESTION_MARK_TRUE: OK
/core/model/part-property/update_connection_designators/QUESTION_MARK_FALSE: OK
/core/model/part-property/update_connection_designators/TILDE_TRUE: OK
/core/model/part-property/update_connection_designators/TILDE_FALSE: OK
/core/model/part-property/update_connection_designators/HASHTAG_TRUE: OK
/core/model/part-property/update_connection_designators/HASHTAG_FALSE: OK
/core/engine/ngspice/watcher/basic: ../src/engines/ngspice-analysis.c:832 @ ngspice_analysis +++ 0 buf = Transient Analysis  Sat Jun 17 10:00:12  2017
**
ERROR:../test/test_engine_ngspice.c:154:test_engine_ngspice_basic: 'distance < 3*16*20' should be TRUE
[1]    17043 abort (core dumped)  /home/bernhard/Projects/oregano/build/src/microtests

```

### Configuration
**Operating system**:
```
cat /etc/os-release |grep PRETTY_NAME
PRETTY_NAME="Fedora 25 (Workstation Edition)"
```

**Versions**
```
oregano --version
Oregano 0.84.5
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster
```
```
ngspice --version

ngspice compiled from ngspice revision 26
Written originally by Berkeley University
Currently maintained by the NGSpice Project

Copyright (C) 1985-1996,  The Regents of the University of California
Copyright (C) 1999-2011,  The NGSpice Project
```

Michi1994, 2017-06-17:
Can you post your `/oregano/test/test-files/test_engine_ngspice_watcher/basic/result/actual.txt` file?

drahnr, 2017-06-17:
Before or after running the test?

drahnr, 2017-06-17:
[actual.txt](https://github.com/drahnr/oregano/files/1082629/actual.txt)

drahnr, 2017-06-18:
The problem is that the output varies between versions of ngspice.

Solutions:

* Do not test the output directly, but the combination of output and parser.
* only check blocks of numerical values

---

Notes:

Unittest should be atomic, file io should be avoided. It might seem to you that reading files is more convenient, but it just adds a lot of overhead and ways to fail compared to a direct include.

drahnr, 2017-06-18:
Closing this in favour of #188

closed 2017-06-18 by drahnr

Michi1994, 2017-06-18:
I will rewrite the assertion part of `ngspice-watcher`-test. I will define a function-metric. The function-metric will measure the distance between two functions that are discrete in time domain. This is necessary, because the time-points of your ngspice are not equal no the time-points of my ngspice. Also the header and footer of our files are too different, so I will extract only the data part like in "ngspice-analysis.c".

drahnr, 2017-06-18:
👍 sounds great