drahnr/oregano #188: Cleanup tests

drahnr, 2017-06-18:
Tests should be atomic and optimally do no file IO.

Michi1994, 2017-06-18:
### Atomic

The problem is, that replacing private functions called in a public function can't be replaced. So if we define "atomic" as "call only one public function", then I agree. But if we define "atomic" as "only one function should be called without any further/deeper function calls" or "only 50 locs at once", then I disagree.

### no file IO

If the out-interface of the function under test is only/includes file IO, then I would say that file IO is okay. In the case of `ngspice-watcher` we ware lucky, because the content of the file is also stored in the ngspice object (in RAM) and we can use that data to assert function equality. BUT I would also like to test the file IO-interface of `ngspice-watcher` so in the end we have to think about how to handle file IO in test cases.

drahnr, 2017-06-18:
**Atomic**

Only one public function was what I meant.

**IO**

Then file IO is still supposedly separate, unless it is really vital to the test itself.

Michi1994, 2017-06-25:
The modified tests of pull request #193 are atomic and file IO is used at a minimum. The tests need file IO because the input and output of the tested function includes file IO.

Michi1994, 2017-06-25:
If someone has further suggestions which tests can be created or how the assertion parts should look like, tell it to me.