drahnr/oregano #189: Can not fork in CI/Docker, makes ngspice tests fail

drahnr, 2017-06-18:
```
gdb --args ./build/src/microtests -p /core/engine/ngspice/watcher/basic
(gdb) r
Starting program: /tmp/build/36d0981b/git-clone-resource/build/src/microtests -p /core/engine/ngspice/watcher/basic
warning: Error disabling address space randomization: Operation not permitted
Cannot create process: Operation not permitted
During startup program exited with code 127.
```

```
$ strace ./build/src/microtests -p /core/engine/ngspice/watcher/basic
strace: ptrace(PTRACE_TRACEME, ...): Operation not permitted
+++ exited with 1 +++
```

drahnr, 2017-06-18:
This seems to require running in privileged mode.

drahnr, 2017-06-18:
Privileged mode does not solve the problem.

Michi1994, 2017-06-18:
There is the old code
```c
static gchar* test_engine_ngspice_get_test_dir_base() {
	gchar *cwd = g_get_current_dir();
	gchar **splitted = g_regex_split_simple("\\/build(?:.(?!\\/build))+$", cwd, 0, 0);
	g_free(cwd);

	gchar *test_dir = g_strdup_printf("%s/test", *splitted);
	g_strfreev(splitted);

	return test_dir;
}
```
in `test_engine_ngspice.c`. The improved code looks like this:
```c
static gchar* test_engine_ngspice_get_test_dir_base() {
	gchar *cwd = g_get_current_dir();

	gchar *test_file = g_strdup_printf("%s/test/test.c", cwd);
	while (!g_file_test(test_file, G_FILE_TEST_EXISTS)) {
		g_free(test_file);
		gchar **splitted = g_regex_split_simple("\\/*(?:.(?!\\/))+$", cwd, 0, 0);
		g_free(cwd);
		cwd = g_strdup(*splitted);
		g_strfreev(splitted);
		test_file = g_strdup_printf("%s/test/test.c", cwd);
	}
	g_free(test_file);
	gchar *test_dir = g_strdup_printf("%s/test", cwd);
	g_free(cwd);

	return test_dir;
}
```
Furthermore I detected a memory allocation/address failure somewhere in the code. It does not fail always, but sometimes. Maybe we should fix that first.

drahnr, 2017-06-18:
What I did so far:

* decouple test exectution from the buildsystem
* briefly look into glib test

what I just started:

* look into running this test in a separate subprocess
https://developer.gnome.org/glib/stable/glib-Testing.html#g-test-trap-subprocess
* test cleanup

drahnr, 2017-06-18:
The function is probably best located in `test.c` itself and rename it to `get_test_dir()` since it could be of use to all testcases.

drahnr, 2017-06-18:
Yes, definitley look into that. I would suggest `valgrind --tool=memcheck --leak-check=full --trace-children=yes --track-origins=yes` to get started.

drahnr, 2017-06-18:
See 3b93607510ab4b8c226c1e228a2bc11bf9935289

drahnr, 2017-06-18:
Discarded the idea of a separate process for this test, would not yield any gains as far as I tell.

Michi1994, 2017-06-18:
It looks like ngspice is working now on the test system, BUT see [https://ci.spearow.io/builds/793](https://ci.spearow.io/builds/793). There you can see the line `Error: Could not find include file /usr/local/share/oregano/models/1N4148.model`.

drahnr, 2017-06-18:
Unfortunately it seems that build just got wiped by the janitor on the CI system..

drahnr, 2017-06-18:
The problem is, at some point in the past ngspice was not installed yet, but now it is https://github.com/drahnr/oregano/commit/80a04427da48e825ea9d1f8aa2dda775e8678215

I actually checked it manually and logged into the CI system.

Michi1994, 2017-06-18:
Is there a variable that describes the `/usr/local/share/oregano/models`-path?

Michi1994, 2017-06-18:
I found it: `OREGANO_MODELDIR`

Michi1994, 2017-06-18:
In [https://ci.spearow.io/teams/main/pipelines/oregano/jobs/pr-compile/builds/4](https://ci.spearow.io/teams/main/pipelines/oregano/jobs/pr-compile/builds/4) you can see that the test is only failing because of #183.

closed 2017-06-23 by drahnr