drahnr/oregano #194: oregano_warning popup - missing check for correct thread

drahnr, 2017-06-18:
Make sure it is used right, and nobody will ever include X11 again.

gtrentalancia, 2017-06-21:
I suppose this issue has been closed with my commit:

https://github.com/gtrentalancia/oregano/commit/8dd76c5c89becdd3340e364dd6f9f222e4552f1a

and your subsequent merge:

https://github.com/drahnr/oregano/commit/4e8c529f22563191978db36bfff654282896ab81

Please let me know if there is anything else I can do. Also, for next time you merge, please include the full ChangeLog for the commit, so it is easier to track this sort of things (the issue resolution was stated in the original commit).

drahnr, 2017-06-21:
No, this is a general problem.

There needs to be a check for those warning/logging funcs if we are in the correct thread and deal with it if not.

Michi1994, 2017-06-25:
I don't understand the problem.

gtrentalancia, 2017-06-25:
The two functions oregano_error() and oregano_warning() should only run in the main thread. As simple as that...

gtrentalancia, 2017-06-25:
Should hopefully be fixed here:

https://github.com/gtrentalancia/oregano/commit/27f932e900740550432e285cacc579f9ca2ac369

drahnr, 2017-06-25:
I was more thinking of `g_thread_self` checks, storing the initial thread ID and comparing all others with it. This would have the advantage to have the platform handling of glib baked in withoud any additional hassles.