drahnr/oregano #198: crash on transient analysis

drahnr, 2017-06-21:
### Steps to reproduce
1. transient analysis
2. 50 points, 2s, 0.1s steps
3.

### Expected behaviour
No crash

### Actual behaviour
Tell us what happens instead

### Configuration
**Operating system**:
```
Fedora 25
```

**Versions**
```
oregano --version
ngspice --version
```

```
(oregano:14909): GLib-WARNING **: GChildWatchSource: Exit status of a child process was requested but ECHILD was received by waitpid(). Most likely the process is ignoring SIGCHLD, or some other thread is invoking waitpid() with a nonpositive first argument; either behavior can break applications that use g_child_watch_add()/g_spawn_sync() either directly or indirectly.
../src/engines/ngspice-analysis.c:216 @ parse_transient_analysis +++ TRANSIENT: result str
>>>
Transient Analysis  Wed Jun 21 06:45:16  2017
<<<

../src/engines/ngspice-analysis.c:587 @ ngspice_parse +++ 1 buf = elapsed time since last call: 0.027 seconds.

../src/engines/ngspice-analysis.c:605 @ ngspice_parse +++ 2 buf = 

(oregano:14909): Gtk-CRITICAL **: gtk_combo_box_text_get_active_text: assertion 'GTK_IS_COMBO_BOX_TEXT (combo_box)' failed

(oregano:14909): GLib-CRITICAL **: g_strsplit: assertion 'string != NULL' failed
[1]    14909 segmentation fault (core dumped)  oregano

```

drahnr, 2017-06-21:
@gtrentalancia can you please look into this?
[error.zip](https://github.com/drahnr/oregano/files/1090362/error.zip)
[image]

drahnr, 2017-06-21:
(the circuit is a dummy which passes all requirements to allow simulation but has no functionality in itself)

drahnr, 2017-06-21:
@Michi1994 the trace also shows that `waitpid` with a watcher is not a good idea, you might want to look into that.

gtrentalancia, 2017-06-21:
I cannot reproduce the crash. It must be a Fedora problem (including Glib/Gtk installed there).

However, here is the output from oregano, there is a Gtk-CRITICAL message, but other than that, no segmentation fault, nothing odd.

Please let me know if there is anything else I can do.

Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
../src/engines/ngspice-analysis.c:1014 @ ngspice_analysis +++ 0 buf = Transient Analysis  Wed Jun 21 14:33:09  2017

(oregano:5751): Gtk-CRITICAL **: gtk_widget_is_drawable: assertion 'GTK_IS_WIDGET (widget)' failed
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.

gtrentalancia, 2017-06-22:
As already explained, I cannot reproduce this issue, but can you try the following:

https://github.com/gtrentalancia/oregano/commit/855bfd55f26e8e6ab84de39daa5e1801c44813f2

It's the only thing I can think of at the moment... Please let me know if it sorts out things.

drahnr, 2017-06-22:
I check that tonight.

drahnr, 2017-06-23:
Fixed. Thanks!

closed 2017-06-23 by drahnr