drahnr/oregano #200: Oregano always prompts for schematic file save on close/quit

gtrentalancia, 2017-06-22:
### Steps to reproduce
1. Start oregano
2. Load a schematic file (on the command line, from recent files or from the file manager)
3. Close or quit the application without making any change to the schematic or simulation settings

### Expected behaviour
Oregano wrongly prompts for schematic file save

### Actual behaviour
Oregano should not prompt for schematic file save on close/quit if the schematic and the simulation settings have not been modified

### Configuration
All OS
All oregano versions including latest git
All engines
```

gtrentalancia, 2017-06-22:
Please merge the following to sort out the issue:

https://github.com/gtrentalancia/oregano/commit/674c2ca1d844cdbfc3776a1afdb12e0db38fb7d4

Thanks.

drahnr, 2017-06-23:
👍

closed 2017-06-23 by drahnr