drahnr/oregano #209: using Debian Stretch and gnucap, exits with a segmentation fault.

rashCoder, 2017-07-03:
### Steps to reproduce
1. start oregano
2.build a simple voltage divider shematic (two resistors and a vdc + ground.
3.When a component is rotated, or when you attempt to simulate the circuit it exits with a segmentation fault.

### Expected behaviour
It would be nice if Oregano would not crash.
### Actual behaviour
Oregano crashes with a segmentation fault (this is the exit dialogue from Konsole terminal.)

### Configuration
**Operating system**: Debian Stretch (9), with KDE.
```
cat /etc/os-release |grep PRETTY_NAME
```

**Versions**
```
oregano --version
Oregano 0.84.7
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster
(this is what I download from this Github.)
gnucap --version
Gnucap 2009.12.07 RCS 26.136
The Gnu Circuit Analysis Package
Never trust any version less than 1.0
Copyright 1982-2009, Albert Davis
Gnucap comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome
to redistribute it under the terms of 
the GNU General Public License, version 3 or later.
See the file "COPYING" for details.
include --version

```

drahnr, 2017-07-09:
Gnucap has not been tested in a while, I'd be happy to see patches being submitted fixing such issues. I'll try to have a look into it, or @contributors could have a look into it

rashCoder, 2017-07-10:
It's odd, but it hasn't crashed while rotating in a while, but it has crashed every time that I try to run a simulation so far. My main beef with Oregano from the Debian repos is that simulation doesn't work. It would be nice if this could be fixed. My solution so far has been to run a netlist and simulate with gnucap in terminal.

drahnr, 2017-07-10:
Did you try to use ngspice?

rashCoder, 2017-07-12:
Sorry about the slow reply. I have not tried NG spice. I should add the contrib repo and try it out later.

rashCoder, 2017-07-12:
Compiled and installed Ngspice today. It does seem to work a whole heck of a lot better with NG spice than Gnucap. The simulations are much better as well. I haven't used NGspice much before.

gtrentalancia, 2017-07-20:
Can you try the latest git tree ? I have made some changes to the gnucap engine and to some component rotation functions.

Please let me know if the crash still occurs with latest git and gnucap. Thanks.

gtrentalancia, 2017-07-20:
However, in general, consider that at the moment it is still preferable to use ngspice or spice3 because they are implemented much better and they support a greater number of analysis types.

I have recently added support for AC, Noise and Fourier analysis with ngspice and also I have added the same support with the original spice3 from UC Berkeley.

Unfortunately, Gnucap at the moment is not fully supported, but, if you have time, you can still try if the situation improves a little with the latest git tree...

gtrentalancia, 2017-07-23:
Unless you provide further feedback, after introducing other fixes and conducting further testing with the gnucap engine, I would consider this issue fixed here:

https://github.com/gtrentalancia/oregano/commit/e51be422de04a7550267b0d64183148e10d4391c

commit 03f304ff (2017-07-23): Merge pull request #215 from gtrentalancia/master

* gnucap parser wrong state, fixes #209
* install splash screen image
closed 2017-07-23 by drahnr