drahnr/oregano #212: ld `gtk_show_uri_on_window ()` missing

drahnr, 2017-07-18:
`gtk_show_uri_on_window ()` is new with gtk+ 3.22 
a) ubuntu has an older gtk+ version
b) ubuntu provides a patched gtk+ version which does not provide this

https://developer.gnome.org/gtk3/stable/gtk3-Filesystem-utilities.html#gtk-show-uri-on-window

https://ci.spearow.io/teams/main/pipelines/oregano/jobs/pr-build-pkg-deb/builds/40

drahnr, 2017-07-18:
```
# pkg-config --modversion "gtk+-3.0"
3.18.9
```

```
NAME="Ubuntu"
VERSION="16.04.2 LTS (Xenial Xerus)"
PRETTY_NAME="Ubuntu 16.04.2 LTS"
```

This implies to either use an if/else macro around this or to bump Ubuntu to 17.04

gtrentalancia, 2017-07-19:
Fixed here:

https://github.com/gtrentalancia/oregano/commit/9dee73babf3c2b83d324be5b2ce7a93e3be2806b

I will now create a pull request for this and another minor improvement.

PR #213 (2017-07-20): Close #212 and check for NULL pointer in wire rotation function
Minor improvements to close #212 and to check for NULL pointer in the wire rotation function.
commit 4cd036b1 (2017-07-20): Merge pull request #213 from gtrentalancia/master

* check for NULL pointer in wire rotation function
* be backwards compat, closes #212
closed 2017-07-20 by drahnr