drahnr/oregano #217: oregano crash

szg0000, 2017-07-30:
### Steps to reproduce
1. start oregano
2. draw a schematic with PNP and NPN transistor
3. click to run simulation

### Expected behaviour
not crash
Tell us what should happen
crash.

```
XXXXXXXXXXXXXXXX
XXXXXXXXXXXXXXXX 0x5613dfee3f80 FALSE
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
*** Error in `oregano': munmap_chunk(): invalid pointer: 0x00005613dd7c05fb ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x70bcb)[0x7f2504e33bcb]
/lib/x86_64-linux-gnu/libc.so.6(+0x76f96)[0x7f2504e39f96]
oregano(+0x7ca8c)[0x5613dd7a9a8c]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4b123)[0x7f250761a123]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x15a)[0x7f25076196aa]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4aa60)[0x7f2507619a60]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7f2507619b0c]
/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0(g_application_run+0x20d)[0x7f250590c72d]
oregano(+0x2dad5)[0x5613dd75aad5]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf1)[0x7f2504de32b1]
oregano(+0x2d8ca)[0x5613dd75a8ca]
======= Memory map: ========
5613dd72d000-5613dd7dd000 r-xp 00000000 08:01 441172                     /usr/local/bin/oregano
5613dd9dd000-5613dd9de000 r--p 000b0000 08:01 441172                     /usr/local/bin/oregano
5613dd9de000-5613dd9eb000 rw-p 000b1000 08:01 441172                     /usr/local/bin/oregano
5613df5ca000-5613e0541000 rw-p 00000000 00:00 0                          [heap]
7f24d4000000-7f24d4022000 rw-p 00000000 00:00 0 
7f24d4022000-7f24d8000000 ---p 00000000 00:00 0 
7f24d8000000-7f24d8022000 rw-p 00000000 00:00 0 
7f24d8022000-7f24dc000000 ---p 00000000 00:00 0 
7f24dc000000-7f24dc022000 rw-p 00000000 00:00 0 
7f24dc022000-7f24e0000000 ---p 00000000 00:00 0 
7f24e0000000-7f24e0022000 rw-p 00000000 00:00 0 
7f24e0022000-7f24e4000000 ---p 00000000 00:00 0 
7f24e4000000-7f24e4022000 rw-p 00000000 00:00 0 
7f24e4022000-7f24e8000000 ---p 00000000 00:00 0 
7f24e9b8b000-7f24e9b8c000 ---p 00000000 00:00 0 
7f24e9b8c000-7f24ea38c000 rw-p 00000000 00:00 0 
7f24ea38c000-7f24ea38d000 ---p 00000000 00:00 0 
7f24ea38d000-7f24eab8d000 rw-p 00000000 00:00 0 
7f24eab8d000-7f24eab8e000 ---p 00000000 00:00 0 
7f24eab8e000-7f24eb38e000 rw-p 00000000 00:00 0 
7f24eb38e000-7f24eb38f000 ---p 00000000 00:00 0 
7f24eb38f000-7f24ebb8f000 rw-p 00000000 00:00 0 
7f24ebb8f000-7f24ebbc6000 r-xp 00000000 08:01 400618                     /usr/lib/x86_64-linux-gnu/libcroco-0.6.so.3.0.1
7f24ebbc6000-7f24ebdc6000 ---p 00037000 08:01 400618                     /usr/lib/x86_64-linux-gnu/libcroco-0.6.so.3.0.1
7f24ebdc6000-7f24ebdc9000 r--p 00037000 08:01 400618                     /usr/lib/x86_64-linux-gnu/libcroco-0.6.so.3.0.1
7f24ebdc9000-7f24ebdca000 rw-p 0003a000 08:01 400618                     /usr/lib/x86_64-linux-gnu/libcroco-0.6.so.3.0.1
7f24ebdca000-7f24ebdff000 r-xp 00000000 08:01 400654                     /usr/lib/x86_64-linux-gnu/librsvg-2.so.2.40.16
7f24ebdff000-7f24ebffe000 ---p 00035000 08:01 400654                     /usr/lib/x86_64-linux-gnu/librsvg-2.so.2.40.16
7f24ebffe000-7f24ebfff000 r--p 00034000 08:01 400654                     /usr/lib/x86_64-linux-gnu/librsvg-2.so.2.40.16
7f24ebfff000-7f24ec000000 rw-p 00035000 08:01 400654                     /usr/lib/x86_64-linux-gnu/librsvg-2.so.2.40.16
7f24ec000000-7f24ec022000 rw-p 00000000 00:00 0 
7f24ec022000-7f24f0000000 ---p 00000000 00:00 0 
7f24f0000000-7f24f0021000 rw-p 00000000 00:00 0 
7f24f0021000-7f24f4000000 ---p 00000000 00:00 0 
7f24f4000000-7f24f4022000 rw-p 00000000 00:00 0 
7f24f4022000-7f24f8000000 ---p 00000000 00:00 0 
7f24f8145000-7f24f8147000 r-xp 00000000 08:01 400656                     /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so
7f24f8147000-7f24f8346000 ---p 00002000 08:01 400656                     /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so
7f24f8346000-7f24f8347000 r--p 00001000 08:01 400656                     /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so
[... 90 lines cut ...]
```

### Actual behaviour
Tell us what happens instead

### Configuration
**Operating system**: Debian 9.1
```
cat /etc/os-release |grep PRETTY_NAME
PRETTY_NAME="Debian GNU/Linux 9 (stretch)"

```

**Versions**
```
oregano --version
Oregano 0.84.10
 Website: https://ahoi.io/projects/oregano
 License: GPLv2/GPLv3
 Main Developer: Bernhard Schuster

ngspice --version
ngspice not installed

```

drahnr, 2017-07-30:
Can you attach your schematic?

gtrentalancia, 2017-08-08:
It happened to me too, while using the gnucap engine. It crashed before displaying the simulation results.

Unfortunately, I don't know how to reproduce it...

gtrentalancia, 2017-08-08:
Might (or might not) be fixed here:

https://github.com/gtrentalancia/oregano/commit/f8d98bd59b19588d53d907b30e3fa5b10565d383

Could you please test and report ?

drahnr, 2017-08-11:
@szg0000 can you please retry at your earliest convenience?

closed 2017-08-11 by drahnr