drahnr/oregano #233: c99 build issue

goett, 2018-05-04:
### Steps to reproduce
1. clone repository from github
	git clone https://github.com/drahnr/oregano.git
2. configure
	cd oregano
	./waf configure	
3. build
	./waf build --release

### Expected behaviour
Configure works as expected: 
./waf configure
Setting top to                           : /home/goett/src/oregano 
Setting out to                           : /home/goett/src/oregano/build 
Checking for 'gcc' (C compiler)          : /usr/bin/gcc 
Checking for program 'glib-genmarshal'   : /usr/bin/glib-genmarshal 
Checking for program 'perl'              : /usr/bin/perl 
Checking for program 'glib-mkenums'      : /usr/bin/glib-mkenums 
Checking for program 'glib-compile-schemas' : /usr/bin/glib-compile-schemas 
Checking for program 'glib-compile-resources' : /usr/bin/glib-compile-resources 
Checking for program 'msgfmt'                 : /usr/bin/msgfmt 
Checking for program 'intltool-merge'         : /usr/bin/intltool-merge 
Checking for header locale.h                  : yes 
Checking for library m                        : yes 
Checking for program 'pkg-config'             : /usr/bin/pkg-config 
Checking for pkg-config version >= '0.26'     : yes 
Checking for 'glib-2.0'                       : yes 
Checking for 'gobject-2.0'                    : yes 
Checking for 'gtk+-3.0'                       : yes 
Checking for 'libxml-2.0'                     : yes 
Checking for 'goocanvas-2.0'                  : yes 
Checking for 'gtksourceview-3.0'              : yes 
Checking for large file support               : yes 
Checking for endianness                       : little 
Checking for inline                           : inline 
Checking for program 'clang-format'           : /usr/bin/clang-format 
Checking for program 'gdb'                    : /usr/bin/gdb 
Checking for program 'nemiver'                : not found 
Checking for program 'valgrind'               : /usr/bin/valgrind 
'configure' finished successfully (0.229s)

run ./waf build --release, expect compilation 
### Actual behaviour
Waf: Entering directory `/home/goett/src/oregano/build'
Variant: ''
[ 8/78] Compiling src/engines/netlist-helper.c
[ 9/78] Compiling src/engines/ngspice-analysis.c
[10/78] Compiling src/engines/ngspice-watcher.c
[11/78] Compiling src/engines/ngspice.c
../src/engines/ngspice-watcher.c: In function ‘add_line_numbers’:
../src/engines/ngspice-watcher.c:145:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; splitted[i+1] != NULL; i++)
  ^
../src/engines/ngspice-watcher.c:145:2: note: use option -std=c99 or -std=gnu99 to compile your code

../src/engines/netlist-helper.c: In function ‘update_schematic’:
../src/engines/netlist-helper.c:357:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (GList *iter = store->parts; iter; iter = iter->next) {
   ^
../src/engines/netlist-helper.c:357:3: note: use option -std=c99 or -std=gnu99 to compile your code

../src/engines/ngspice-analysis.c: In function ‘ngspice_table_destroy’:
../src/engines/ngspice-analysis.c:422:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (int i = 0; i < len; i++) {
   ^
../src/engines/ngspice-analysis.c:422:3: note: use option -std=c99 or -std=gnu99 to compile your code
../src/engines/ngspice-analysis.c: In function ‘ngspice_table_new_columns’:
../src/engines/ngspice-analysis.c:470:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (gchar **variable = variables; *variable != NULL && **variable != '\n' && **variable != 0; variable++) {
  ^
../src/engines/ngspice-analysis.c: In function ‘ngspice_table_add_data’:
../src/engines/ngspice-analysis.c:497:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; data[i] != NULL && data[i][0] != 0 && data[i][0] != '\n' && i < ngspice_table->current_variables->len; i++) {
  ^
../src/engines/ngspice-analysis.c: In function ‘get_real_len’:
../src/engines/ngspice-analysis.c:518:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (guint i = 0; i < array_guint->len; i++) {
  ^
../src/engines/ngspice-analysis.c:520:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (guint j = 0; j < i; j++) {
   ^
../src/engines/ngspice-analysis.c: In function ‘parse_transient_analysis_resources’:
../src/engines/ngspice-analysis.c:697:11: error: conflicting types for ‘i’
  for (int i = 0; i < nodes_nb + 1; i++) {
           ^
../src/engines/ngspice-analysis.c:590:8: note: previous definition of ‘i’ was here
  guint i = 0;
        ^
../src/engines/ngspice-analysis.c:697:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; i < nodes_nb + 1; i++) {
  ^
../src/engines/ngspice-analysis.c: In function ‘ngspice_save’:
../src/engines/ngspice-analysis.c:982:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; (pipe = thread_pipe_pop(pipe, &buf, &size)) != NULL; i++) {
  ^
../src/engines/ngspice-analysis.c: In function ‘parse_no_of_variables’:
../src/engines/ngspice-analysis.c:1026:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; i < 5; i++)
  ^
../src/engines/ngspice-analysis.c: In function ‘ngspice_analysis’:
../src/engines/ngspice-analysis.c:1135:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (int i = 0; transient_enabled || fourier_enabled || dc_enabled || ac_enabled || noise_enabled; i++) {
  ^

Waf: Leaving directory `/home/goett/src/oregano/build'
Build failed
 -> task in 'shared_objects' failed with exit status 1 (run with -v to display more information)
 -> task in 'shared_objects' failed with exit status 1 (run with -v to display more information)
 -> task in 'shared_objects' failed with exit status 1 (run with -v to display more information)

Cannot locate proper location to enable c flags. appending '-std=c99' to CFLAGS in wscript not working.

### Configuration
Scientific Linux 7.4 (Nitrogen)

**Versions**
```
oregano not yet installed 
ngspice revision 27
```

drahnr, 2018-05-07:
Thanks for reporting, can you by chance add your compiler version?

drahnr, 2018-05-07:
@goett the following should work for you:
```
export CFLAGS='-std=c99' && ./waf distclean && ./waf --verbose configure build --release
```

Alternatively:
Add another py list item to the follow lines (depends on debug and release build)
https://github.com/drahnr/oregano/blob/master/wscript#L86-L91

which has the same effect.

I can not reproduce the issue, running Fedora 28 `gcc (GCC) 8.0.1 20180324 (Red Hat 8.0.1-0.20)`

Cheers

drahnr, 2018-05-13:
Closing, feel free to re-open if this does not solve your problem.

closed 2018-05-13 by drahnr