drahnr/oregano #239: Wire-drawing pen captures the screen even when the pen is outside of the window

yurivict, 2019-11-23:
This makes the system unusable.

drahnr, 2019-11-24:
#240 should address your issue, given it is only when drawing a wire.

Could you please list all circumstances (wire creation, rubberband, item placement, ...) and your window manager (gnome-shell, kwin, i3, ...) in which you experience the issue. The fix above is only for drawing/creating new wires.

yurivict, 2019-11-24:
> Could you please list all circumstances <...>

I only encountered this once while drawing a wire.

My window manager is XFCE4.

drahnr, 2019-11-24:
How did you obtain you instance of oregano, I'd be happy to give you a pre-release build to verify this is enough.

yurivict, 2019-11-24:
I installed it from the FreeBSD port.

drahnr, 2019-11-24:
I never made a release for, so it'd be awesome if you could give the above a shot :)

yurivict, 2019-11-24:
FreeBSD packages are just built from the ports, either by the central servers or by users themselves, it doesn't matter.

So software maintainers don't need to "make releases" for FreeBSD. Once the port is available the software packages normally also become available.

Please go ahead and make a release with a fix, and I will check it once I update the port.

Thank you!

drahnr, 2019-11-27:
Should be fixed now.

closed 2019-11-27 by drahnr