drahnr/oregano #246: Loudspeaker resistor value not generated in netlist (gnucap tested)

be1, 2020-10-25:
### Steps to reproduce
0. Configure oregano for GnuCAP
1. Open the proof oregano file given in [speaker.zip](https://github.com/drahnr/oregano/files/5435415/speaker.zip) archive
2. Go to menus and display netlist

### Expected behaviour
R_L1 (speaker) line should be : R_L1 1 0 8 ohm

### Actual behaviour
R_L1 line is R_L1 1 0 (missing 8 ohm)

### Configuration
**Ubuntu GNU/Linux 20.10**:
```
PRETTY_NAME="Ubuntu 20.10"
```

**Versions**
```
Oregano 0.84.41
gnucap         1:0.36~20171003-1build2
no ngspice present
```

drahnr, 2020-10-30:
@be1 thanks for reporting, unfortunately I won't have any capacity for oregano anymore these days ( see #161 ) but I'd be happy to guide you through the source if you want to tackle it yourself!

be1, 2020-10-30:
@drahnr thanks for your reply. I'll see if I can fix the code, on this week-end.

be1, 2020-10-31:
I think I found the bug in the xml libary. But this leads to another issue: the speaker model. Indeed, a speaker is in fact not a resistor, neither an inductor, neither the combination of the two. Because of the mechanical parts, It's more or less a combination of a R-L and a R-L-C networks, the first in series, and the second in parrallel. But even that model is wrong because the R-L part is not a good model. manufacturers use this as a starting point to design their crossover filters and adjust it after crossover application measures.

I suggest anyway to use R as the model for the speaker, because speaker manufacturer does not give the R-L/R-L-C specifications. Are you okay with that ? For people having a good knowledge of it, they can themselves create a compound of R-L / R-L-C if they want their own description of their loudspeaker.

be1, 2020-10-31:
The generated netlist is now correct. See attachement oregano file.
[speaker.zip](https://github.com/drahnr/oregano/files/5468319/speaker.zip)

commit 5411dffd (2020-11-01): Fix #246 bad speaker definiton
closed 2020-11-01 by drahnr